openeew/openeew-firmware #7: Implement buzzer sounds

johnwalicki, 2021-04-01:
Currently the buzzer will sound an earthquate alert sound if a shake is detected or if the device receives a MQTT message from the regional network declaring an alarm.

Not exactly "music" but we can make the sensor beep different status messages too.

johnwalicki, 2021-04-01:
The gotcha is that esp-idf does not implement the arduino tone() function 
https://esp32.com/viewtopic.php?f=13&t=16839&sid=fe47120147b9794023ba8d59e07c1aa9

johnwalicki, 2021-04-01:
The relevant esp-idf docs are here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html

johnwalicki, 2021-04-01:
Do we want to tag this for the v1.5.0 firmware and Caribbean deployment?
We might push this to the backlog if it's not in the critical path.

neilh10, 2021-04-01:
What if the audible is tied to the system states, https://github.com/openeew/openeew-firmware/issues/7    that would define what states need an audible component and keep it integrated with the major visual status.  It simplifies testing.

When a device/product is successful it is often because there are user interface components that are configurable - the tones - so potentially (for the future) an extensible framework that could allow tone configuration. (ouch sorry seems like just needs one pleasant sound for the first pass !)   

One of the issues with audible is a user "cut-off"  ~  "thanks I heard, now turn off". 
This was standard requirement for all telecom equipment that had an Infiniti of notifications for the technicians to be alerted. 

The cut-off was both messaging, the virtual  (MQTT?),   and also the physical in front of the device. 
Could a "double tap" on the side of the device be detected,(and separated out from earthquake readings)  for an audible cutoff.

andygrillo, 2021-04-01:
I really like the idea of the double tap !

I think we would need to run that by Vaclav and get seismology input first!

Please note that the buzzer isnt very loud, and should likely only be on
during an earthquake when its necessary and people are preparing for
shaking.