openeew/openeew-firmware #10: Heap Fragmentation problem

johnwalicki, 2021-04-13:
The firmware uses ArduinoJson and over several days/weeks might be affected by heap fragmentation

```
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
CORRUPT HEAP: Bad head at 0x3ffe436c. Expected 0xabba1234 got 0x3ffed498
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x40135e57 on core 1
```

johnwalicki, 2021-04-13:
This issue has been fixed in the version150 branch.

johnwalicki, 2021-04-13:
The firmware has been modified to remove several DynamicJsonDocument globals that could cause heap fragmentation because they were malloc/freed every second in the main loop.  The variables have moved into the Send MQTT data functions that are called much much less frequently.

Long investigation thread in the Slack channel #firmware

neilh10, 2021-04-13:
I checked out the https://github.com/openeew/openeew-firmware/compare/version150 - but its way beyond me at this point except for the version string :).
Would it work to have an "alpha" branch to put these into and use some of the semavers.org version which would allow auto updates  for testing if needed.  That is the version would be called 1.5.0-alpha01 until its merged with the (main)?

johnwalicki, 2021-04-14:
If you're interested in reviewing the code, but not worrying about compiling it,  I've got a test firmware-1.5.0.bin on the update server. We can flip a switch and force push a firmware upgrade to your device.

neilh10, 2021-04-14:
sure happy to. you know my number errrr mac   :)  That would test the upgrade path.

johnwalicki, 2021-04-14:
I've implemented a Firmware Beta allowlist.  I've added your device mac address to the allowlist.  The next time you restart the device (or poke it remotely by sending it a Firmware Check MQTT command from the Control dashboard), it will do a firmware upgrade to v1.5.0 beta.

@andygrillo  You might be interested in this feature too.  We talked about a few months ago.  Let's us control which devices get which firmware.  Its a brute force comparison for specific mac addresses - not particularly scalable.

johnwalicki, 2021-04-14:
Fixed via https://github.com/openeew/openeew-firmware/commit/19aabb6bec5e6535280c9dda7b000cd7e0c257cf

neilh10, 2021-04-14:
Hey worked for me very nicely. I power cycled.

Tue Apr 13 23:09:48 2021

Contacting the OpenEEW Device Activation Endpoint :
https://device-mgmt.openeew.com/activation?ver=1
Sending Device Activation : {"macaddress":"A8032A4DCCCC","firmware_device":"1.4.0"}
HTTP Response code: 200
HTTP post response payload: {"org":"7p7xwk","firmware_latest":"1.5.0","firmware_ota_url":"https://openeew-devicemgmt.mybluemix.net/firmware/firmware-1.5.0.bin"}
OpenEEW Device Activation directs MQTT data from this sensor to :7p7xwk
Version 1.5.0 is higher than: 1.4.0
An OTA upgrade is required. Download the new OpenEEW firmware :
https://openeew-devicemgmt.mybluemix.net/firmware/firmware-1.5.0.bin
LED_FIRMWARE_OTA - Magenta
Opening Server PEM Chain : /mybluemix-net-chain.pem
Read /mybluemix-net-chain.pem pem server certificate chain from SPIFFS
Starting OpenEEW OTA firmware upgrade...
OTA upgrade downloaded. Restarting...
WiFi Stopped
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

OpenEEW Sensor Application
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started

neilh10, 2021-04-14:
Plugged in the ethernet and let it run.

ahhh a CORRUPT HEAP -  good ole Backtrace has a story, but who can read it 

Sending 1 second of accelerometer readings in a MQTT packet of size: 668
{"d":{"traces":[{"x":[0.049,0.151,0.079,0.174,0.155,0.143,0.143,0.143,0.219,0.106,0.185,0.087,0.109,0.14,0.132,0.204,0.117,0.132,0.147,0.128,0.147,0.14,0.223,0.196,0.174,0.091,0.045,0.117,0.049,0.132,-0.023,0.091],"y":[-0.385,-0.219,-0.223,-0.215,-0.223,-0.275,-0.313,-0.256,-0.177,-0.366,-0.287,-0.328,-0.29,-0.328,-0.317,-0.366,-0.343,-0.343,-0.283,-0.14,-0.253,-0.298,-0.343,-0.287,-0.204,-0.336,-0.339,-0.381,-0.268,-0.162,-0.279,-0.419],"z":[1.999,2.025,2.014,2.124,1.988,2.052,2.112,2.063,2.059,1.984,2.135,1.992,2.131,2.059,2.025,2.048,2.052,2.176,2.003,2.007,2.048,1.992,2.131,2.01,2.127,2.131,2.127,2.059,1.935,2.074,2.12,2.041]}],"device_id":"A8032A4DCCCC"}}
Sending 1 second of accelerometer readings in a MQTT packet of size: 668
ETH Disconnected
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
CORRUPT HEAP: Bad head at 0x3ffe84cc. Expected 0xabba1234 got 0x3ffed9d0
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x40135ce3 on core 1

Backtrace: 0x4008d6dc:0x3ffb9220 0x4008d90d:0x3ffb9240 0x40135ce3:0x3ffb9260 0x4008d351:0x3ffb9290 0x40084cc2:0x3ffb92b0 0x40084399:0x3ffb92d0 0x400f5232:0x3ffb92f0 0x400ee232:0x3ffb9310 0x400d9974:0x3ffb9330 0x400d94d5:0x3ffb9350 0x400d6ef5:0x3ffb9370 0x400d2ef6:0x3ffb9390 0x400d8896:0x3ffb93d0 0x400d8965:0x3ffb94c0 0x40089dc9:0x3ffb94f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

OpenEEW Sensor Application
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started
Completed scan for access points
WiFi Network scan done
9 network(s) found
1: Arthur2004Sid (-35)*
..
ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
Found network Arthur2004Sid , xxxxxx
Disconnected from WiFi access point
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
There was a problem connecting to WiFi
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Found no matches for saved networks
ESP32 WiFi interface ready
ESP32 WiFi started
Unhandled Network Interface event : 14
Waiting for SmartConfig.
ESP32 soft-AP stop
ESP32 soft-AP stop
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue

Even with ethernet, got stuck in "Blue".   required power cycle. I do see this every other power cycle. Should I write a bug on just "not coming up after power cycle?"   I'll try timing my power cycle to make sure I have it off for 10seconds and see if still has a problem.

Next power cycle with ethernet plugged in  It came back OK

johnwalicki, 2021-04-14:
Two things going on in the log.
- Before the heap corruption, there's a ETH disconnect.  That might be a clue.
```
Sending 1 second of accelerometer readings in a MQTT packet of size: 668
ETH Disconnected
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
CORRUPT HEAP: Bad head at 0x3ffe84cc. Expected 0xabba1234 got 0x3ffed9d0
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
```

- After the power cycle, it reads NVM ram to get your WiFi password, tries to connect for 60 seconds and fails.
```
ESP32 WiFi started
Found network Arthur2004Sid , xxxxxx
Disconnected from WiFi access point
LED_LISTEN_WIFI - Blue
...
There was a problem connecting to WiFi
Got no match for network
...
Found no matches for saved networks
<enters the SmartConfig loop>
```

johnwalicki, 2021-04-14:
While the code certainly is not behaving correctly, the network failures on both Ethernet and WiFi seem odd. Is there some other problem on your network that could be contributing?

johnwalicki, 2021-04-14:
I suspect that the WiFiClientSecure and underlying ssl_client library are not handling Ethernet connects / disconnects well.  I can hotplug / unplug the Ethernet cable a few times successfully. Then, it crashes.   Below, the log shows a ETH disconnect with a successful WiFi/MQTT connect, ETH hotplug with a successful ETH/MQTT connect, a ETH disconnect with a successful WiFi/MQTT connect, ETH hotplug with a successful ETH/MQTT connect, then a crash on the ETH disconnect.  The log doesn't show timestamps but I waited a few seconds between each event.

Internet searches on these errors find numerous reports over the past 2 years related to this.  ESP32 libraries are just not mature.

```
ETH Disconnected
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:D5:F3
ETH MAC: A8:03:2A:4D:D5:F3, IPv4: 192.168.1.198, FULL_DUPLEX, 100Mbps
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Disconnected
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:D5:F3
ETH MAC: A8:03:2A:4D:D5:F3, IPv4: 192.168.1.198, FULL_DUPLEX, 100Mbps
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
CORRUPT HEAP: Bad head at 0x3ffe84cc. Expected 0xabba1234 got 0x3ffffff8
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x40135ce3 on core 1
```

johnwalicki, 2021-04-14:
There's a section of the firmware that handles network events.  It detects Ethernet events and WiFi events.  On Ethernet hotplug/unplug events, the firmware will try to disconnect MQTT from the prior interface and switch to the new interface.  Seems rock solid.  I just added the Serial.println() statements. I can do this over and over (dozens of times)
```
    case SYSTEM_EVENT_ETH_GOT_IP:
      eth_connected = true;

      // Switch the MQTT connection to Ethernet from WiFi (or initially)
      // Preference the Ethernet wired interence if its available
      // Disconnect the MQTT session
      if( mqtt.connected() ){
        Serial.println("Previously connected to WiFi, try to switch the MQTT connection to ETH");
        mqtt.disconnect();
        // No need to call mqtt.setClient(ETH); because ETH is a ETHClient which is not the same class as WiFi client
        // Connect2MQTTbroker(); // The MQTT reconnect will be handled by the main loop()
      }
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      // Disconnect the MQTT client
      if( mqtt.connected() ){
        Serial.println("Previously connected to ETH, try to switch the MQTT connection to WiFi");
        mqtt.disconnect();
      }
      break;
```

johnwalicki, 2021-04-14:
The 004C errors are somewhat expected (the socket closed abruptly)
`#define MBEDTLS_ERR_NET_RECV_FAILED                       -0x004C  /**< Reading information from the socket failed. */`

Here's a small snip of the log as I hotplug/unplug ethernet.
```
ETH Disconnected
Previously connected to ETH, try to switch the MQTT connection to WiFi
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:D5:F3
ETH MAC: A8:03:2A:4D:D5:F3, IPv4: 192.168.1.198, FULL_DUPLEX, 100Mbps
Previously connected to WiFi, try to switch the MQTT connection to ETH
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Disconnected
Previously connected to ETH, try to switch the MQTT connection to WiFi
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:D5:F3
ETH MAC: A8:03:2A:4D:D5:F3, IPv4: 192.168.1.198, FULL_DUPLEX, 100Mbps
Previously connected to WiFi, try to switch the MQTT connection to ETH
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():281]: (-76) UNKNOWN ERROR CODE (004C)
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
```

neilh10, 2021-04-14:
Thanks for the detail... 
my network ...   a Synology RT2600AC router for both wired and wireless.  It also does wirespeed Layer 4& 7 management  bad URL cop. https://www.synology.com/en-us/products/RT2600ac#specs
It is dual WAN, so have comcast cable at 29MPBS and Sonic/ATT DSL at 4MBPS.

If there could be an underlying WiFi network problem should we spin this off to a sperate issue? 
I've always found it best to figure out network issues in as simple an environment as possible.   
I had an issue with a seeedstudio WIO TERMINAL WiFi (https://www.seeedstudio.com/Wio-Terminal-p-4509.html)  last year when they released it. 
I had written a test routine to learn the API and bang on the basics, and it failed so I posted it. They admitted that there was a rewrite going on, so I shelved the project.  https://forum.seeedstudio.com/t/stable-wifi-ntp-failing/254153 
I haven't been able to get back to that project yet

johnwalicki, 2021-04-14:
My firmware version that prints the extra debug seems to be very stable for me. Didn't add anything else.
```
Previously connected to WiFi, try to switch the MQTT connection to ETH
...
Previously connected to ETH, try to switch the MQTT connection to WiFi
```

johnwalicki, 2021-04-15:
I added some extra Ethernet / WiFi connectivity reporting in this commit:
https://github.com/openeew/openeew-firmware/commit/94098f1036fcdc6c7b6967dfa9393dcc528c9439
I have NOT seen a heap fragmentation error.

johnwalicki, 2021-05-01:
I'm absolutely confident that moving the mqtt.disconnect() on network interface changes into the Connect2MQTTbroker() will remove any asynchronous problems that were causing heap corruption.
Fixed by https://github.com/openeew/openeew-firmware/commit/6d1b031b2b6d4d811e982f428c962e68af528332

johnwalicki, 2021-05-01:
Closed by https://github.com/openeew/openeew-firmware/pull/25

closed 2021-05-01 by johnwalicki