openeew/openeew-firmware #16: Change hostname from openeew-sensor-wifi to openeew-sensor-macaddr

johnwalicki, 2021-04-28:
it might be nice to have a more informative string "OpenEEW-<last 6digits MAC>
The firmware currently sets the hostname to "openeew-sensor-eth" or "openeew-sensor-wifi"
If you do an scan on your local network, you'll find the openeew-sensor device.
It would be easy to add the MAC address to the name.

This is handled in the `NetworkEvent()` function.

johnwalicki, 2021-04-28:
WiFi was setting the hostname in the wrong spot.  Fix in https://github.com/openeew/openeew-firmware/commit/5b773a7adf604884215429f30489eb74420503d9

neilh10, 2021-04-29:
That worked for me on my router DHCP Clients showing as "openeew-sensor-wifi"

closed 2021-05-01 by johnwalicki

neilh10, 2021-05-01:
Well per the title - this should have some unique device ID at the end of the host name, some of the last digits (6) from the MAC
Per a build I did from V1.5.0 this is what my router shows - no unique ID

[image]

neilh10, 2021-05-01:
I think this issue should be re-opened.