openeew/openeew-firmware #17: heap corruption on ethernet line fast changes

neilh10, 2021-04-28:
With my device WiFi Max4D CC CC with downloaded code from branch version150  (in process) had some heap Corruptions. 
This appears to be disconnects/connects on the physical line, there was no manual action.  
The ethernet connects to a D-Link DSS16+ 16port switch.  I usually use this link with a raspeberry Pi and its never had any problems.
I've posted it here, because of the heap corruptions, but it does recover with a reboot.
There may be other causes for the disconnects/connects 

CORRUPT HEAP: Bad head at 0x3ffe8cac.
probably same as before .. but just in case
LED_CONNECTED - Cyan
{"d":{"traces":[{"x":[0.06,0.068,0.079,0.083,0.011,-0.049,0.008,0.075,0.023,0.03,0.091,0.06,0.117,0.038,0.049,0.068,0.102,0.109,0.072,-0.004,0.075,0.057,0.128,0.166,0.023,0.034,0.091,0.091,0.015,0.049,0.019,0.091],"y":[-0.03,0.008,0.034,-0.049,-0.079,-0.087,-0.011,0.087,-0.079,-0.094,-0.064,0.041,0.038,0.106,0.008,-0.181,-0.083,0.045,0.041,-0.117,-0.143,0.004,0.075,-0.128,-0.106,-0.109,-0.106,-0.045,0.068,-0.109,-0.147,-0.053],"z":[2.312,2.199,2.282,2.32,2.267,2.24,2.437,2.263,2.301,2.263,2.357,2.293,2.301,2.323,2.395,2.38,2.263,2.222,2.278,2.278,2.301,2.312,2.342,2.421,2.342,2.376,2.342,2.259,2.274,2.256,2.267,2.256]}],"device_id":"A8032A4DCCCC"}}
Sending 1 second of accelerometer readings in a MQTT packet of size: 656
LED_CONNECTED - Cyan
ETH Disconnected
Previously connected to Ethernet, try to switch the MQTT connection to WiFi
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-76) UNKNOWN ERROR CODE (004C)
CORRUPT HEAP: Bad head at 0x3ffe8cac. Expected 0xabba1234 got 0x3ffecfa0
abort() was called at PC 0x40086fe9 on core 1
ELF file SHA256: 0000000000000000
Backtrace: 0x40088f08:0x3ffb8f60 0x40089185:0x3ffb8f80 0x40086fe9:0x3ffb8fa0 0x40087115:0x3ffb8fd0 0x400fc643:0x3ffb8ff0 0x400f8705:0x3ffb92b0 0x400f869d:0x3ffb9300 0x4008e37d:0x3ffb9330 0x40081e46:0x3ffb9350 0x40081fb5:0x3ffb9370 0x4012f5b6:0x3ffb9390 0x4012297a:0x3ffb93b0 0x400d8d48:0x3ffb93d0 0x400d88a9:0x3ffb93f0 0x400d6115:0x3ffb9410 0x400d1d1b:0x3ffb9430 0x400d7b56:0x3ffb9470 0x400d7c60:0x3ffb9560 0x4008ac4a:0x3ffb9590
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
OpenEEW Sensor Application
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started
Completed scan for access points
WiFi Network scan done
11 network(s) found
1: SonicAdslSsid (-43)*
2: Arthur2004Sid (-49)*
..

ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
Found network Arthur2004Sid , xxxxxx
Got no match for network
LED_LISTEN_WIFI - Blue
ESP32 WiFi connected to AP
ESP32 station got IP from connected AP
Obtained IP address: 10.66.66.110
LED_LISTEN_WIFI - Blue
WiFi was successfully connected
WiFi Connected
WiFi MAC: A8:03:2A:4D:CC:CC
ETH  MAC: A8:03:2A:4D:CC:CF
A8032A4DCCCC
ETH Started
Waiting for time
Time sync'd
Thu Apr 22 00:47:40 2021

johnwalicki, 2021-04-28:
In one of the traces, it showed the ethernet connecting/disconnecting many times.  Eventually there's a heap corruption.
For certain its a bug (I think in the underlying espressif libraries) I'm not certain what to do here.
I suspect if we solve the connect/disconnect issue, we would not see the heap corruption failures.

Possibly some faulty hardware and Ethernet wiring?
Would be interesting to swap around your three devices if its reproducable on the others?

neilh10, 2021-04-29:
Will try some mix and match of the equipment to see if it changes anything
Because I captured it, maybe its useful, this was the section ...
Fifo read number 20
Retrieved 32 entries
Total/Average X=-2079566/-3249; Y=2525513/3946; Z=165818709/259091
ADXL355 Accelerometer activated
ADXL355 Accelerometer first samples discarded
Finished accelerometer configuration
<<<seems there was a long wait here>>>>
ETH Disconnected
Previously connected to Ethernet, try to switch the MQTT connection to WiFi
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
Previously connected to WiFi, try to switch the MQTT connection to Ethernet
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Disconnected
Previously connected to Ethernet, try to switch the MQTT connection to WiFi
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-76) UNKNOWN ERROR CODE (004C)
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
Previously connected to WiFi, try to switch the MQTT connection to Ethernet
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Disconnected
Previously connected to Ethernet, try to switch the MQTT connection to WiFi
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
MQTT Connected
ETH Disconnected
Previously connected to Ethernet, try to switch the MQTT connection to WiFi
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-76) UNKNOWN ERROR CODE (004C)
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
MQTT Connected
ETH Disconnected
Previously connected to Ethernet, try to switch the MQTT connection to WiFi
Attempting MQTT connection...LED_CONNECT_CLOUD - Cyan
MQTT Connected
ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
Previously connected to WiFi, try to switch the MQTT connection to Ethernet
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-76) UNKNOWN ERROR CODE (004C)
CORRUPT HEAP: Bad head at 0x3ffe84cc. Expected 0xabba1234 got 0x3ffeec58
abort() was called at PC 0x40086fe9 on core 1
ELF file SHA256: 0000000000000000
Backtrace: 0x40088f08:0x3ffb8f60 0x40089185:0x3ffb8f80 0x40086fe9:0x3ffb8fa0 0x40087115:0x3ffb8fd0 0x400fc643:0x3ffb8ff0 0x400f8705:0x3ffb92b0 0x400f869d:0x3ffb9300 0x4008e37d:0x3ffb9330 0x40081e46:0x3ffb9350 0x40081fb5:0x3ffb9370 0x4012f5b6:0x3ffb9390 0x4012297a:0x3ffb93b0 0x400d8d48:0x3ffb93d0 0x400d88a9:0x3ffb93f0 0x400d6115:0x3ffb9410 0x400d1d1b:0x3ffb9430 0x400d7b56:0x3ffb9470 0x400d7c60:0x3ffb9560 0x4008ac4a:0x3ffb9590
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
OpenEEW Sensor Application
ESP32 WiFi interface ready

closed 2021-05-01 by johnwalicki