openeew/openeew-firmware #18: List MACs on power up

neilh10, 2021-04-28:
Idea:  When doing a power up, it would be useful to list the MAC addresses in a human readable form  
This could be used by anybody, or a manufacturing machine, from the terminal when the system powers up. 
Currently 1.4.0 the MACs are only listed out following the configuration.
If somebody needed to find the MACs before configuration, they can't.

LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.Completed scan for access points
WiFi Network scan done
10 network(s) found
1: AzondeNetSsid (-44)*
....
ESP32 WiFi interface ready
Found no matches for saved networks
ESP32 WiFi started
ESP32 WiFi interface ready
WiFi Stopped
Waiting for SmartConfig.
LED_WIFI_OFF - White
ESP32 soft-AP stop
ESP32 soft-AP stop
ESP32 soft-AP stop
ESP32 soft-AP stop
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue

then when configured through the App

.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
ESP32 WiFi connected to AP
LED_CONNECT_WIFI - Green
LED_CONNECT_WIFI - Green
LED_CONNECT_WIFI - Green
ESP32 station got IP from connected AP
Obtained IP address: 192.168.1.172
LED_CONNECT_WIFI - Green
SmartConfig received.
Waiting for WiFi
Smart Config done, connected to: AzondeNetSsid with psswd: xxxxxx
Writing network to NVM: AzondeNetSsid,xxxx
Stored networks in NVM: 0
Device has 1 networks stored in NVM
LED_CONNECT_WIFI - Green
WiFi Connected
WiFi MAC: A8:03:2A:4D:CC:B4
ETH  MAC: A8:03:2A:4D:CC:B7
A8032A4DCCB4
ETH Started

johnwalicki, 2021-04-28:
The firmware does list the mac addresses (eventually and if you troll through the console log).
Very few (non-geeky) people are going to connect a serial monitor to the board.
This request is to print them sooner?

Or would you prefer to have the mac address as part of the hostname?
See this ticket: https://github.com/openeew/openeew-firmware/issues/16

neilh10, 2021-04-28:
Yes the request is to print them sooner if possible. 

My idea is that is to be able to uniquely ID the device easily, possibly before they have a serial number printed on them, and could be used in the unit assembly process by @andygrillo manufacturing program.
Once @andygrillo manufacturing program has loaded the production firmware, at the factory, it won't get configured in the factory, so listing them earlier would allow the ID to be visible.
@andygrillo this is what I was thinking of. Production firmware download for unit assembly by your rPi test jig, that would list the MACs.  Would it be useful  for your rPi manufacturing program in the unit assembly to be able to parse the MAC at this point,  and then print them to  any label that is created for the openEEW sensor unit before packaging.   

Is it easier to have a JSON like ID to be able to pull them off in your manufacturing program?  

It would also be a type of manufacturing test, as if able to parse a valid MAC it probably indicates a lot of the openEEW sensor hardware is working.

johnwalicki, 2021-05-01:
I left this GitHub issue open because there are other requests here.

neilh10, 2021-05-02:
For the latest download from openeew-firmware-main (main)  version 1.5.0 "git pull;   Updating 0884667..9336cd6"
the MAC numbers don't show before it gets a configuration.
For no WiFi networks active (but ethernet plugged in),  I can't see what the MACs are, -

I was thinking power up is before configuration.

OpenEEW Sensor Application
ESP32 WiFi interface ready
Stored networks : 2
ESP32 WiFi started
Completed scan for access points
WiFi Network scan done
8 network(s) found
1: Kitchen speaker.o (-66)
2: PurpleAir-42fc (-67)
3: LazySky (-67)*
4: ATT6HzT3GE (-81)*
5: AzondeNetSsid (-90)*
6: DIRECT-57-HP OfficeJet 3830 (-90)*
7: HOME-B202 (-92)*
8: HomeBaseWifi104 (-94)*
ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
Found network Arthur2004Sid , xxxxxx
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Reading stored networks from NVM
Found network ArthurGuestSsid , xxxxxx
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Found no matches for saved networks
ESP32 WiFi interface ready
WiFi Stopped
Waiting for SmartConfig.
LED_WIFI_OFF - White
ESP32 soft-AP stop
ESP32 soft-AP stop
ESP32 soft-AP stop
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
.LED_LISTEN_WIFI - Blue
..... 

FOR a WIFI network ArthurGuestSsid present and Ethernet plugged in

johnwalicki, 2021-05-03:
You have some other build.  The log above isn't the from the final 1.5.0 release.
In the scenario of a registered device but that WiFi network not available, with an Ethernet cable connected, the log should look like this:
```
OpenEEW Sensor Application
ETH Connected
ETH MAC: A8:03:2A:4D:D5:F3, IPv4: 192.168.1.186, FULL_DUPLEX, 100Mbps
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started
Completed scan for access points
WiFi Network scan done
5 network(s) found
1: wynd565o (-45)*
2: wynd565o (-55)*
3: wynd565o (-81)*
4: Fios-Gk6Rn (-90)*
5: oliver (-93)*
ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
Found network 565guest , xxxxxx
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Found no matches for saved networks
Stored networks : 1
Previously registered device, use hardwired Ethernet connection.
WiFi MAC: A8:03:2A:4D:D5:F0
```

neilh10, 2021-05-03:
Many thanks. Don't know what happened with my build and download- driver error I guess - still learning it. 

Now showing
OpenEEW Sensor Application
ETH Started
ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
ESP32 WiFi interface ready
Stored networks : 2
ESP32 WiFi started
ETH Disconnected

I'm closing this as that was the purpose to have a unique ID.

closed 2021-05-03 by neilh10