openeew/openeew-firmware #19: openEEW Sensor sometimes doesn't boot

neilh10, 2021-04-29:
On plugging in my device with a Version150 it sometimes won't boot up.
It continues to try, however I have to unplug, wait a few seconds, then re-plug it in. 
Sometimes I have to do this a couple of times before it boots.

The USB is plugged into either an Ubuntu  20.04LTS running through a self powered hub, or a Windows 10 through its own self powered USB hub.

rst:0x3 (SW_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jun  8 2016 00:22:57

neilh10, 2021-04-29:
With a terminal running, on Windows and plugging the device in the boot looks like this 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

OpenEEW Sensor Application
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started

andygrillo, 2021-04-29:
I am having a similar issue. After ADXL configuration is stops:

Total/Average X=1379722/2155; Y=162417056/253776; Z=-9311712/-14549
ADXL355 Accelerometer activated
ADXL355 Accelerometer first samples discarded
Finished accelerometer configuration

Yet, if I unplug ethernet, it works and starts sending.

johnwalicki, 2021-04-30:
These crash loops seem to occurring before the firmware even gets control.  Yet other times when the ESP32 powers up, it works fine?   
``` rst:0x3 (SW_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)```
https://github.com/espressif/esp-idf/blob/release/v3.0/components/esp32/include/rom/rtc.h#L80

johnwalicki, 2021-05-01:
See LED brownout discussion in https://github.com/openeew/openeew-firmware/issues/23
I believe this issue will be fixed by https://github.com/openeew/openeew-firmware/commit/50cac2b5ec3719eaa5aca7b0aad1563a103dca4f

closed 2021-05-01 by johnwalicki

neilh10, 2021-05-01:
Hey a lot of late night coding. Thanks so much.
It worked on my device CCCC.
Fingers crossed