openeew/openeew-firmware #20: sensor doesn't always connect over ethernet

neilh10, 2021-04-29:
With a build from the version150 branch, and only ethernet the sensor doesn't always find the ethernet.

I provision the sensor xx 4D CC CC on a WiFi network and it works. 

If I plug in the Ethernet it connects over this ethernet.
Then I remove the WiFI and repower it, it comes up switches to the Ethernet and uses it.
Using a Windows 10 with a powered USB hub, this is what the trace looks like:

```
OpenEEW Sensor Application
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started
Completed scan for access points
WiFi Network scan done
8 network(s) found
1: SonicAdslSsid (-47)*
2: LazySky (-73)*
3: AzondeNetSsid (-81)*
....
ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
Found network Arthur2004Sid , xxxxxx
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Found no matches for saved networks
Stored networks : 1
Use hardwired Ethernet connection.
WiFi MAC: A8:03:2A:4D:CC:CC
ETH  MAC: ETH Started
A8:03:2A:4D:CC:CF
A8032A4DCCCC

Waiting for time
LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.ETH Connected
ETH MAC: A8:03:2A:4D:CC:CF
ETH MAC: A8:03:2A:4D:CC:CF, IPv4: 10.66.66.113, FULL_DUPLEX, 100Mbps
LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.Time sync'd
Thu Apr 29 14:26:17 2021

Contacting the OpenEEW Device Activation Endpoint :
```

Then I move it to another location, where the orginal provisioned WiFi network is minimal signal strength, and different Ethernet hub,  it doesn't seem to find the ethernet.
Using an Ubuntu 20.04LTS  with a powered USB hub, this is what the trace looks like:

```
Completed scan for access points
WiFi Network scan done
10 network(s) found
1: AzondeNetSsid (-32)*
2: Lars (-80)*
3: Lars (-86)*
4: DIRECT-58-HP DeskJet 3700 series (-86)*
5: SonicAdslSsid (-86)*
6: Terri (-88)*
7: HomeBaseWifi104 (-89)*
8: Arthur2004Sid (-90)*    **<<<< provisioned Network>>>>>**
9: LazySky (-95)*
10: F089 (-96)*
ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
Found network Arthur2004Sid , xxxxxx
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
Got no match for network
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
LED_LISTEN_WIFI - Blue
Disconnected from WiFi access point
LED_LISTEN_WIFI - Blue
There was a problem connecting to WiFi
Got no match for network
Got no match for network
Found no matches for saved networks
Stored networks : 1
Use hardwired Ethernet connection.
WiFi MAC: A8:03:2A:4D:CC:CC
ETH  MAC: A8:03:2A:4D:CC:CF
A8032A4DCCCC
ETH Started

Waiting for time
LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.Disconnected from WiFi access point
LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.Disconnected from WiFi access point
LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
.LED_FIRMWARE_DFU - Yellow
```
Repeated by re-powering the hub a number of times,  
Full trace attached
[ttyUSB2_2104291054.txt](https://github.com/openeew/openeew-firmware/files/6400874/ttyUSB2_2104291054.txt)

johnwalicki, 2021-04-30:
Hmm - This trace file is a jumble.  I can't make sense of it.  It logs output from many different sections of code.

johnwalicki, 2021-05-01:
I believe this issue will be resolved via https://github.com/openeew/openeew-firmware/issues/22
and this patch https://github.com/openeew/openeew-firmware/commit/83a69fce932248b87c8ee5724dcbe7f91d0971e2

closed 2021-05-01 by johnwalicki