openeew/openeew-firmware #29: Adopt the ESPNtpClient library to get millisecond NTP accuracy

johnwalicki, 2021-06-16:
Make sure NTP is as accurate as possible by using the `ESPNtpClient` library that could help us get to millisecond accuracy:
https://github.com/gmag11/ESPNtpClient 

The ESPNtpClient library is available in PlatformIO.  It might be easy to incorporate into the firmware.
I could be added to the `SetTimeESP32()` function.

```
NTP.onNTPSyncEvent ([] (NTPEvent_t event) {
        ntpEvent = event;
        syncEventTriggered = true;
    });
```
We'll have to study the example to see how to merge it into the firmware.

andygrillo, 2021-06-16:
If possible move the NTP function over to the 2nd core of the ESP32:
https://randomnerdtutorials.com/esp32-dual-core-arduino-ide/

eg,
```

TaskHandle_t Task1;
TaskHandle_t Task2;

// LED pins
const int led1 = 2;
const int led2 = 4;

void setup() {
  Serial.begin(115200); 
  pinMode(led1, OUTPUT);
  pinMode(led2, OUTPUT);

  //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
  delay(500); 

  //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */
    delay(500); 
}

//Task1code: blinks an LED every 1000 ms
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    digitalWrite(led1, HIGH);
    delay(1000);
    digitalWrite(led1, LOW);
    delay(1000);
  } 
}

//Task2code: blinks an LED every 700 ms
void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    digitalWrite(led2, HIGH);
    delay(700);
    digitalWrite(led2, LOW);
    delay(700);
  }
}

void loop() {
  
}
```

DhruvaG2000, 2021-09-23:
I just tested the example on the ESP32 and the output is as below: 
[image]

(I am guessing it works, well)

I went through the source code and it seems do-able to me to add it to this project.
I have also some experience with FreeRTOS in ESP32 hence I think we can pin this task to core, however I can't promise on that because wifi in itself is a resource heavy task and running anything on the same core on which wifi is running does lead to some ``GURU MEDITATION Errors`` in my experience. 

But it surely is worth a shot. I will need some help though getting acquainted with this project's code base. Can someone point as to where I can add this NTP Sync stuff in this repo?

johnwalicki, 2021-09-23:
There is a function called SetTimeESP32() in main.cpp
It is called at ESP32 powerup / initialization and then once a day.

DhruvaG2000, 2021-09-23:
I suppose you mean this: https://github.com/openeew/openeew-firmware/blob/main/AWS_IoT/src/main.cpp#L779-L807 ?

DhruvaG2000, 2021-09-23:
So the changes IMO might be something like:

1. First, in setup we need to add:
```C
    NTP.onNTPSyncEvent ([] (NTPEvent_t event) {
        ntpEvent = event;
        syncEventTriggered = true;
    });
```
2. then modify the existing function as
```C
void SetTimeESP32() {
        NTP.setTimeZone (TZ_Europe_Madrid); // depends on what we want
        NTP.setInterval (600);
        NTP.setNTPTimeout (NTP_TIMEOUT);
        // NTP.setMinSyncAccuracy (750);
        // NTP.settimeSyncThreshold (500);
        NTP.begin (ntpServer);
     
// then to show the time serially, or however, something like below:
Serial.print (NTP.getTimeDateStringUs ()); Serial.print (" ");
}
```

DhruvaG2000, 2021-09-23:
Also please tell me what should the directory struct be like, my current understanding is that:
```
.
|- include (https://github.com/openeew/openeew-firmware/tree/main/AWS_IoT/include)
     |- ESPNtpClient.cpp (https://github.com/gmag11/ESPNtpClient/tree/main/src)
     |- ESPNtpClient.h  (Add the folder I mentiooned above as a submodule maybe?)
     |- SetTimeESP32.cpp
     |- SetTimeESP32.h

```

johnwalicki, 2021-09-23:
Yes, sounds good.  I've assigned this ticket to you. 
I would maybe keep
https://github.com/openeew/openeew-firmware/blob/main/AWS_IoT/src/main.cpp#L780-L782
and
https://github.com/openeew/openeew-firmware/blob/main/AWS_IoT/src/main.cpp#L803-L805

In setup(), immediately after the SetTimeESP32(), the function creates the MQTT connection.  If the time is wrong, the MQTT connect will fail.  That's why I wait the additional second here:
https://github.com/openeew/openeew-firmware/blob/main/AWS_IoT/src/main.cpp#L801

When you submit a PR, I'll also implement it in the WatsonIoT/src/main.cpp 
Or you can submit two commits in the PR to update both.

DhruvaG2000, 2021-09-23:
as far as pinning to the second core goes, the pseudo syntax is:

```C
TaskHandle_t Task1;
xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */                  
void Task1code( void * pvParameters ){
// our time sync code
}
```

DhruvaG2000, 2021-10-12:
@johnwalicki and @krook can we close this issue ?

closed 2021-10-12 by johnwalicki